FRCTeam3206/Electrical #12: Systems / Electrical plan through school start Fall 2016

RajGandhi, 2016-05-14:
For through summer:

Monday (as the basis), two hour meetings (taking into account learning experiences along the way).

May 16th- Discuss thoughts for wiring and layout for next year, start work on the controller and work with breadboards

May 23rd- Continue work on controller board

May 30th- Wrap up work on controller board (?) - play with pneumatic capabilities

--meet on other day, possibly programming, in order to complete controller board

June 3rd- Possible Programming collaboration with arduino project

Skip June 6th due to it being finals (?)(could be nice to get in extra day, but attendance would be hard)

June 13th- Testing of controller board, arduino projects

June 17th- Meet on programming day to finish arduino projects if needed (or wanted)

---- June 18th-August 1st 

I send super cool emails to do with systems-online resources to familiarize with wiring and components, maybe to make into a really fun format so people will actually read it- and encourage attendance at programming and mechanical meetings.
 
----- 

August 2nd- Practice design layouts for electrical components, work with some older components and extras to practice wiring, make a project out of it--a shooter mechanism, something involving wiring motors

August 8th- People make cool stuff

August 15th- Wrap up and talk about upcoming year

(Possibility to space out previous meetings more or add more content)

Crimping training included when there is extra time, or too many people working on one thing
-----

School Starts on the 6th- are we going to have a current whole team meeting after the school year starts to discuss plan and how to recruit people?

I maybe got a little too specific, but I thought as the end of the school year is approaching having a more solid idea of where and when to start would be good.  If more time is needed to prepare for the controller, then some independent pneumatics could get interesting or just having a blast with breadboards. Let me know what you think and how your schedule works with this rough sketch.

Arduino projects- probably start off with some basics and then try to have designs with more purpose-like the turret

Daniel-Thalman, 2016-05-14:
So that is why you have been there on mondays

RajGandhi, 2016-05-14:
Yes but not exactly causal.  The plan is from Jessica.  Apparently her schedule is better on Mondays than most days.

Last week we received the club's RFID board and it seemed like a good idea to have it soldered and ready to go for Thursday to support programming.  Monday was basically the only/best day to get the work done.

For upcoming Monday, 5/16, planning to meet 6-7:30pm.

closed 2016-11-01 by RajGandhi