FRCTeam3206/Electrical #16: Team Laptop

RajGandhi, 2016-05-31:
The existing team drive-station laptop may not be adequate for next season.  It was upgraded to Windows 10 and now LabView takes a _long time_ to do ordinary tasks.

Where is the touch-screen laptop and what is that status?

It is probably a good idea to start the search process for the next laptop, just in case it is necessary.  
Needs:
- Runs latest LabView
- Runs latest Eclipse & Java
- Runs Autodesk Inventor 2015 (that's our standard, right?)
- Upgraded to Windows 10 

Wants:
- 16 Gb system RAM (8 Gb might be okay but is there much of cost difference?)
- 500 Gb SSD (250 Gb SSD might be  adequate; spinning hard drive is slower and less reliable)
- 15" Screen (is 17" too big for driver station and 13" too small for development?)
- video co-processor with 2 Gb video ram
- Windows 10 native (upgrading to Windows 10 is a hassle)

Raj's big laptop is a MacBook Pro 15 2013 that dual-boots to Windows 7.  The hardware specs are good and it runs LabView, Eclipse and Inventor fine.  Cost is too high and the Windows OS lacks refinement (i.e. the drivers aren't well tuned).  There are shut-down issues and wake-ups will sometimes lose the keyboard.  

Without any justification, here's a link to the [MacBook Pro 15 2015 review](http://www.notebookcheck.net/Apple-MacBook-Pro-Retina-15-Mid-2015-Review.144402.0.html).  More importantly is the concluding paragraph that mentions competitors:
If you are still undecided between the MacBook and the Windows alternatives like the [Asus ZenBook Pro](http://www.notebookcheck.net/Asus-ZenBook-Pro-UX501JW-Notebook-Review.142308.0.html), the [Acer Aspire V 15 Nitro](http://www.notebookcheck.net/Acer-Aspire-V15-Nitro-Black-Edition-VN7-591G-Notebook-Review.140919.0.html) or the [Asus G501JW](http://www.notebookcheck.net/Asus-G501JW-Notebook-Review.141745.0.html), which have been available for quite a while now... The battle for the first spot in our [Top 10 ranking](http://www.notebookcheck.net/Notebookcheck-s-Top-10-Multimedia-Notebooks.98608.0.html) remains exciting.

Chief Delphi [thread 1](http://www.chiefdelphi.com/forums/showthread.php?t=148135) and [thread 2](http://www.chiefdelphi.com/forums/showthread.php?t=140733).

Autodesk Inventor [certified hardware search tool](http://usa.autodesk.com/adsk/servlet/syscert?id=18844534&siteID=123112).  All of these results include Quadro video cards and, as a result push the price too high, $2k or so.  Carolyn would like a 2nd laptop capable of Inventor.  The "certified hardware" is probably more than adequate and can't justify the extra $1k.  Anything with a video co-processor is probably going to be adequate.  Raj's laptop has a [NVIDIA GeForce GT 750M](http://www.notebookcheck.net/NVIDIA-GeForce-GT-650M.90245.0.html) that is known to be adequate.

It appears that the Acer line might be the best value.  Something that meets all the Needs and some of the Wants should be available in the $600 to $900 range.  Consider purchasing options including Best Buy, NewEgg, Amazon.

It would be nice to find someone with personal experience with one of the (Acer) laptops.  If a good choice can be found, the team can make the recommendation to the greater team and possibly buy two for the team.

Daniel-Thalman, 2016-05-31:
For development screen size dosn't matter ie i use a 11 inch air and the screen is no hinderence

RajGandhi, 2016-06-03:
Not specifically a recommendation, just research notes.

Asus ROG GL 552 gaming laptop [review](http://www.laptopmag.com/reviews/laptops/asus-rog-gl552#)
Asus [site](https://www.asus.com/ROG-Republic-Of-Gamers/ROG-GL552JX/Features/); about $1000 on NewEgg
Graphics [Nvidia GeForce GTX 960M](http://www.notebookcheck.net/NVIDIA-GeForce-GTX-960M.138006.0.html) 2GB of VRAM  

[Cheaper version](http://www.newegg.com/Product/Product.aspx?Item=N82E16834232946) on NewEgg, $679
Asus [site](https://www.asus.com/Notebooks/K501UX/)
Graphics [NVIDIA GeForce GTX 950M](http://www.notebookcheck.net/NVIDIA-GeForce-GTX-950M.138026.0.html)

RajGandhi, 2016-06-07:
Existing team laptop is [Lenovo U530](http://www.notebookcheck.net/Review-Lenovo-IdeaPad-U530-Touch-Notebook.115269.0.html)

According to the specs it is pretty good: i7, dual core (four logical cores), dual video processors (Nvidia GT 730M), 8Gb, 1 Tb HDD with 600+ Gb free.

RajGandhi, 2016-10-06:
Roger, can you make a recommendation based on some of the above notes?

Daniel-Thalman, 2016-10-06:
My 2013 13 inch MacBook Air dual boots to Win 10 and has no more issues then a native Win 10 but also include extra cost of Windows in price. Which is about $100

RajGandhi, 2016-11-04:
Purchased.

closed 2016-11-04 by RajGandhi